open-rdc/ROS2_Walking_Pattern_Generator #6: 動歩行の達成

Yusuke-Yamasaki-555, 2022-10-15:


Yusuke-Yamasaki-555, 2022-11-10:
静歩行は一旦白紙。逆運動学->動歩行を考える。

Yusuke-Yamasaki-555, 2023-06-13:
本格的な動歩行の実装に向けて、開発を勧めていく。

[ROBOTIS OP2の重量・足裏から腰までの高さ。](http://en.robotis.com/img/sub/pruduct_img_OP2_07.png)
・重量：3.0[kg]
・腰の高さ：219.5[mm]

これを元に、軌道エネルギーの式を立てて、歩行パラメータを構築する。そしたら、歩行パターン生成のための実装を進めていく。

実装方法は、はじめは動けば良いので、WebotsRobotHandlerに直接実装する。step()でリアルタイムに生成するのではなく、init()で歩行パターンを全て計算しておき、step()で順次パターンを読んで動かす方法を取る。

Yusuke-Yamasaki-555, 2023-06-23:
理論のそれっぽいコードは書けたので、まずは重心軌道がちゃんと生成されているか確認している。
まだ上手くできていない。

本の数式と、本と先生のコードのアルゴリズムを参考に構築中。コードとアルゴリズムに関しては、コピペはしてないし、写してもいない、見ながらもやっていないので、セーフ。数式はほぼ写しているが、コレは別問題。

Yusuke-Yamasaki-555, 2023-06-24:
プロットしてみたが、やっぱしうまくいっていない。用修正。

[image]

Yusuke-Yamasaki-555, 2023-06-24:
単に、着地位置がだめかも？特にはじめの２，３歩。計算がおかしいのだろうな。

Yusuke-Yamasaki-555, 2023-06-24:
[image]

ce5e28362d617b446130d7ba85d70e98d2f4a733

記録

Yusuke-Yamasaki-555, 2023-06-24:
改めて考え直したら、色々と腑に落とせた。アルゴリズムは理解できたような気がする。コレならちゃんとしたプログラムが書けそう。
上で挙げたグラフとかは、やっぱ理論の実装が色々と間違ってた。

基準座標のx軸が右足の足裏中心点を通るのは少し気持ち悪いが、IKを解くときやグラフに直すときに、y軸要素の値を調整すれば良いか。

[image]

Yusuke-Yamasaki-555, 2023-06-25:
着地位置の修正をしないとこんな感じ。なんかできてそうなできてなさそうな。

[image]

Yusuke-Yamasaki-555, 2023-06-25:
それっぽいグラフは得られた。足踏みの重心軌道はうまく得られている気がする。ただ、初めの1,2歩は思惑通りにいっていない。

まだ釈然としないというか、負に落とせているとは思うが、あまりうまくいかんなぁ。XとYでそれほど大きな違いは無いはずなんだが。

e29a13fcc38a6a5be8e77d8a4dbc7a1b60d9d75b

[image]

Yusuke-Yamasaki-555, 2023-06-26:
X軸方向の運動も入れてみた。んーそれっぽい。
教科書のやつとY軸が結構違うのは、脚の長さや着地位置が違うからかな。

[image]

やっぱ、X軸を考える場合は歩行パラメータをちゃんと定めないと、ありえない値が得られちゃうな。

[image]

Y軸のものであっても、歩行パラメータを変えてやったり最適化問題の重みパラメータを変えてやれば、もっといい感じの結果が得られそうだなと思った。

Yusuke-Yamasaki-555, 2023-06-27:
いやー、単位が違いました。

着地位置を[m]で扱っていたのに、脚の長さを[mm]で扱っていました。

Yusuke-Yamasaki-555, 2023-06-27:
?????

[image]

Yusuke-Yamasaki-555, 2023-06-27:
教科書のパラメータでPlotしてみた。ちゃんとしたものが出ている。

[image]

　まず、初期重心位置がだめだった。初期支持脚着地位置と初期重心位置を一致させなければ、教科書の方法はだめな気がする。
　他は特に問題がなかったので、あとはパラメータの設定次第。

Yusuke-Yamasaki-555, 2023-06-27:
初期姿勢は直立が良いので、もう１ステップ増やして初期着地位置を直立状態の重心位置としてやればいい感じになるかもしれない。

Yusuke-Yamasaki-555, 2023-06-27:
倒立伸子なんだし、初期支持脚位置と初期重心位置がズレてたら、そら豪快に倒れるわな。

2足歩行ロボットじゃなくて倒立伸子であることを意識しなきゃダメ。

だから、最初は重心と支持脚が同じ位置にするために、直立状態と同じ支持脚位置を初期として、やるべきなんじゃないか？

Yusuke-Yamasaki-555, 2023-06-27:
いや、やっぱ上手くいっていないな。初めの1、2歩がやっぱダメだな

Yusuke-Yamasaki-555, 2023-06-27:
やっぱ１，２歩目がだめだった。初期着地位置に対して、着地位置修正を掛けていなかったのが原因だった。

これでちゃんと教科書の動きが再現できたはず。

[image]

Yusuke-Yamasaki-555, 2023-06-27:
足踏みで速度もプロットしてみたVersion。めっちゃきれい。最高。

[image]

Yusuke-Yamasaki-555, 2023-06-27:
前進に速度もプロットしてみたVersion。本当は別のグラフに分けたほうが良いが、ひとまずこれで。

[image]

Yusuke-Yamasaki-555, 2023-06-27:
開発環境に合わせて、足踏みの歩行パラメータにしてみたVersion。あとはパラメータ調整して、遊脚軌道（正弦波）考えて、IK解いて、ロボットに適用すればOKかな。

[image]

Yusuke-Yamasaki-555, 2023-06-27:
初期重心位置を直立状態に合わせたVersion。修正着地位置が重心位置と一致したので、これは実現不可能ですね。

歩き始めはやっぱ、こちらから右脚の片足支持まで持っていってから歩き始めたほうがいいのか？

[image]

Yusuke-Yamasaki-555, 2023-06-28:
静歩行の実装の時に使った方法を参考に実装するのも良いだろう。

Yusuke-Yamasaki-555, 2023-06-28:
初期重心位置は別に直立状態で良いか。着地位置もそれに合わせよう。直立状態は両脚支持なんだから、その時の着地位置は実質的に体の中心にある。

Yusuke-Yamasaki-555, 2023-06-28:
最適化問題の重みを変えてみた。１、１０、１００と変えてみたが、ほとんど変化なし。グラフにはほとんど現れていないので、省略。

まあもともと、着地位置の修正が見られなかったので、重みを変えてもほとんど変化が無いのは自明の理、的な。

なので、変えるとしたら着地位置ではあるのだが、今の着地位置は足を上下にしか動かすだけになっている。これ以上内側に設定したら、左右の足が切替時に干渉しかねないので、これで行くしかないかなぁと思っている。

Yusuke-Yamasaki-555, 2023-06-28:
次にやること。

：遊脚軌道の生成と、実装方法の検討

Yusuke-Yamasaki-555, 2023-07-05:
歩行パラメータを生成して実装するところまでを、ひとまずやってみた。

が、やはりそううまくは動かなかった。

以下、Webots側で出力されたエラーの一部
```
DLegLowerR HingeJoint > Solid > DEF DAnkleR HingeJoint > RotationalMotor: too low requested position: -2.72454 < -1.24
WARNING: DEF ROBOTIS_OP2 RobotisOp2 (PROTO) > DEF DPelvYR HingeJoint > Solid > DEF DPelvR HingeJoint > Solid > DEF DLegUpperR HingeJoint > Solid > DEF DLegLowerR HingeJoint > Solid > DEF DAnkleR HingeJoint > Solid > DEF DFootR HingeJoint > RotationalMotor: too low requested position: -3.0882 < -0.68
WARNING: DEF ROBOTIS_OP2 RobotisOp2 (PROTO) > DEF DPelvYL HingeJoint > RotationalMotor: too big requested position: 3.14159 > 2.5
WARNING: DEF ROBOTIS_OP2 RobotisOp2 (PROTO) > DEF DPelvYL HingeJoint > Solid > DEF DPelvL HingeJoint > Solid > DEF DLegUpperL HingeJoint > Solid > DEF DLegLowerL HingeJoint > Solid > DEF DAnkleL HingeJoint > RotationalMotor: too big requested position: 2.80885 > 1.22
```

Yusuke-Yamasaki-555, 2023-07-05:
歩行パラメータで生成した指定関節角度がだめ。指定範囲外の値になっている。

以降、原因を特定して、バグを修正する。

Yusuke-Yamasaki-555, 2023-07-05:
* PelvYL: 左脚腰関節Z軸: 足踏み・直進では使わない。3.14159（つまり180度）しか与えていない。つまりIKの問題？
* AnkleL: 左脚足首Y軸
* FootL: 左脚足首X軸: この２つはよくわからん。どっちにしろIKが悪さしてそう。

Yusuke-Yamasaki-555, 2023-07-05:
遊脚軌道は正弦波を考えているわけだが、その適用方法は、重心位置のZ軸から正弦波の値をそのまま引いてやった重心位置を、遊脚がIKを解くときに使えは、遊脚軌道が達成できると考える。

IKは各脚の床面との設置点を基準点（0, 0, 0）としているので、重心位置から引いてやる方法しかないとは思うが。

以下、重心位置をプロットした３次元グラフ。単にプロットしただけだから分かりづらい。
[image]

このグラフ的に、ちゃんと重心位置から遊脚軌道が引かれていて、支持脚切替タイミングで元の重心位置と一致しているので、OKだと思われ。初めのX軸0~0.8では遊脚が発生しないので無視で。（両脚支持期間はまだ設けてない。）

Yusuke-Yamasaki-555, 2023-07-05:
やっぱIKか？

Yusuke-Yamasaki-555, 2023-07-05:
それより、他の遊脚軌道の実装例を見たほうが良いか

Yusuke-Yamasaki-555, 2023-07-05:
てかそもそも、脚が届かない可能性もあるのか。ちゃんと計算していないが。

Yusuke-Yamasaki-555, 2023-07-13:
やること
・コードの再確認、バグを出している箇所の特定。
・座標系の基準を、足裏床面から腰位置に変更してやってみる。
・歩行パターンからＩＫを解いた結果が、脚の可動範囲内にあるかの確認。

Yusuke-Yamasaki-555, 2023-07-15:
それっぽい動歩行の動きはできた。歩行パターンは[上記](https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/issues/6#issuecomment-1610920695)で示した足踏みのものを使用。

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/7226e0f5-ccd6-4e02-a3d8-62aa54d6276e

足の軌道はちゃんとしているはず。以下は、重心位置に対する足までの距離。重心軌道はZMPに近づくように円弧を描くので、支持脚の足と重心位置までの距離が円弧に沿って縮まったりするのは正しいはず。逆に遊脚の足が重心位置から円弧に沿って離れていったりするのも正しいはず。遊脚の足が正弦波の軌道で上がっているのも確認できる。このグラフは一般的なものでは無いはずだからわかりにくいと思われる。一応記録。
[image]

ただ、動画を見る限りキレイな足踏みにはなっていない。
* 両脚支持期間の実装ができていないから？
* 歩行パラメータが駄目だから？
* まだ足の軌道とIKがうまく適用できていない？
* 初めの0.8[s]の足の軌道がおかしいから？

10[ms]の制御周期は、WebotsのbaseTimeStepを10[ms]に設定することで適用している。今回は通信を挟んでいないので、webotsのstep()関数内で歩行パラメータを読み込んで各関節に適用しているだけなので、問題になるオーバーヘッドは存在しないはず。

Yusuke-Yamasaki-555, 2023-07-17:
動歩行で足踏みができているっぽい。でも思ったとおりの動きにはなっていない（特に遊脚を数cm上げてほしいのに上がっていない）し転倒しちゃっているっぽいので、パラメータとか足先位置とかを確認する。

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/822628a9-a49f-4bbd-97e2-7fdeb2a0586f

Yusuke-Yamasaki-555, 2023-07-25:
今は、歩行パターンから支持脚・遊脚の関節角度を与えるコードを書いている。が、特にX方向が上手くいっていない（集中して取り組んでいないのも原因）。
以下、X軸方向における足の位置の時間変化。0.02前に足を置くように指定しているのだが、0.03前までも足を伸ばして0.01までしか足を引いていない。式が駄目だねぇ。
コードもぐちゃぐちゃになってきたし、一旦整理し直そうか。

[image]

参考に、今の歩行パターン。Debug用に色んな軌道を出しているので見せられるレベルのものではないが。

[image]

Yusuke-Yamasaki-555, 2023-07-25:
今の歩行パラメータ。着地位置はWorld座標系で、右足裏を基準(0, 0)として設定。

{{0.0, 0.0, 0.037},  // 歩行パラメータからの着地位置(time, x, y)
  {0.8, 0.0, 0.074},  // 元は、0.037. 基準点を変えている. 
  {1.6, 0.02, 0.0}, 
  {2.4, 0.04, 0.074},  
  {3.2, 0.06, 0.0},
  {4.0, 0.08, 0.074},
  {4.8, 0.08, 0.037},
  {5.6, 0.08, 0.037}};

Yusuke-Yamasaki-555, 2023-07-26:
X軸方向はOKかな。ただ、歩行終了時に倒れてしまうので、Y軸を修正する必要があるっぽい。

[image]

Yusuke-Yamasaki-555, 2023-07-26:
んんんーーーーー

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/d8ac4f4e-0dd5-4735-bc36-8cd3f1b58553

[image]

[image]
[image]
[WSC2WRH_clock.log](https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/files/12171387/WSC2WRH_clock.log)

Yusuke-Yamasaki-555, 2023-07-26:
歩行パラメータを変えてみた。X軸方向を、0.05[m]に変更。

どうもかかとを引きづっているっぽい。

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/3289e4e6-9809-4cf9-913d-af904c8b4909

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/e718ddef-8109-4a6c-bcbb-e168affd028d

Yusuke-Yamasaki-555, 2023-07-26:
まあ、動歩行は成功してますねコレは

Yusuke-Yamasaki-555, 2023-07-26:
遊脚の正弦波のコードがアレだったので、修正。

Yusuke-Yamasaki-555, 2023-07-26:
うん、ここまでできればまず良いだろ。動歩行ひとまずOK、達成！

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/01f63463-f0ee-4fb6-a705-7cb7757d9887

yasuohayashibara, 2023-07-27:
まだちょっと足を引きずっているように見えます．
遊脚の足を上げる量を増やしてみて下さい．

Yusuke-Yamasaki-555, 2023-07-27:
ありがとうございます。もう少し調整してみます。

Yusuke-Yamasaki-555, 2023-07-28:
現状：

遊脚軌道：高さ0.08[m]の正弦波+0.01[m]。
両脚支持期間：0.2[s]。歩行開始時と終了時に0.1[s]の両脚支持。

足を引きずっている。特に踵が床を擦っている。コレの原因はわからん。足首Pitchに直接角度を加えるか、IKや速度指令がよろしくないのか。
あと、両脚支持をIKのときに適用しているため、歩行パターンとの乖離が多少あるのかもしれない。教科書では歩行パターン生成時に両脚支持期間を適用していたので、そこも要因か？

まずは、両脚支持期間を歩行パターン生成にちゃんと組み込むところから始めるべきか。

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/f2ffd6f8-6593-41b5-88b2-812b03acd52c

[image]
[image]
[image]
[WSC2WRH_clock.log](https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/files/12192575/WSC2WRH_clock.log)

Yusuke-Yamasaki-555, 2023-07-28:
ああいや、違うか？両脚支持の適用方法を復習する。

Yusuke-Yamasaki-555, 2023-07-29:
遊脚や両脚支持期の脚の軌道を調整してみた。ほとんど足を引きずらずに歩けるようになった。
しかし、歩行パラメータをいじると全く歩けなくなるので、やはりまだ遊脚軌道が駄目っぽい。前に指摘してもらっていた最大速度で動かすことをしていないからかも。実際、今回遊脚の初めと終わりに0.06[m]足を一瞬で上げ下げするようにしているので、実質最大速度なのだと思う。
あと、両脚支持期の時に、重心の速度・加速度の変化を滑らかにするような実装をしていないのも原因の１つかも。

以下、記録。

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/c08d9acd-f83c-4d7d-b703-721937fa0317

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/6605c643-dd9c-4374-8e1d-4a2ccb0fcc2c

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/5498f521-18ca-42ce-8991-1ae4fe918f73

胴体中心投影点に対する脚の軌道
[image]
遊脚軌道
[image]
歩行パターン
[image]
胴体中心投影点に対するX軸方向の脚の軌道
[image]

Yusuke-Yamasaki-555, 2023-07-29:
速度を最大にしてみようか。もうそれぐらいしか分からん。

Yusuke-Yamasaki-555, 2023-07-29:
速度を最大にしてみた。遊脚軌道を、正弦波（高さ0.08[m]）に変更。まあちゃんと歩けそうなんだけど、やっぱ何故か足裏が床面と並行になっている。そのせいでつま先で床面を蹴ってしまって、歩けていないんだろうなと。コレどうしたものか。

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/66b8887c-f937-4508-8f1d-b19012b8c547

Yusuke-Yamasaki-555, 2023-07-29:
速度を限定的に最大にしてみた。

確かにいい感じ！　足は引きずっていない。ただ、着地時のタイミングなのか、滑るときがある。

遊脚を上げ下げする、初めの0.05[s]だけ、脚各関節最大速度(12.26[m/s])を適用。

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/127809d1-71b4-4ca8-98cc-c3e62a577332

yasuohayashibara, 2023-07-29:
歩行の目標軌道を見ると，地面から8cmまで持ち上げるようなパターンになっています．
動画を見るとそれほど足を上げていないため，おそらく逆運動学の問題のような気がします．

Yusuke-Yamasaki-555, 2023-07-30:
ありがとうございます。確認してみます。

Yusuke-Yamasaki-555, 2023-08-02:
足が80mmほどまで上がっていない問題、IKとFKを、直立状態のリンク状態から計算しているから説。

Yusuke-Yamasaki-555, 2023-08-02:
そうか、既に膝を曲げて中腰っぽくなっているから、直立状態と現在の腰高さ差分をとって、差分＋足上げ高さで正弦波を計算してやる必要があるか。

Yusuke-Yamasaki-555, 2023-08-02:
腰を基準にした
* 直立高さ：0.186 [m]
* 歩行時高さ：0.171836 [m]
* 高さ差分：0.014164 [m]

高さ差分＋足上げ高さを正弦波の振り幅に適用するので、今の振り幅0.08だと、足上げ高さは
0.08 - 0.014164 = 0.065836 [m] ~= 66 [mm]
となっているはず。

Yusuke-Yamasaki-555, 2023-08-02:
んー、でも確かに66[mm]持ち上がっている感じでもなさそう。IKを確認します。

https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/assets/91410662/cf074021-3b54-4697-a7af-53d4cc21b2ff

Yusuke-Yamasaki-555, 2023-08-03:
んー、逆運動学は問題なさそう。
OP2の脚構造は教科書の解析的手法の構造とほぼ一致しているので、ほぼ教科書のママ実装しているわけで。比較してみたが、特におかしな点はなかった。

ただ１点、p66一番下の姿勢の式に関しては、移行が間違っている気がしているので、そこは変更している。右辺のRx, Ryをinnverse()している。実際、教科書のママだとおかしなことになるし。

よくわからん。
足首Yaw軸から足裏面までのリンク長をIKで考えていないからか？。いや腰〜足首まででちゃんと足上げができていれば別に問題は無いはずなんだが。

たまに挙げる以下のグラフはIKから得た関節角度を元にFKを解いた結果で、ちゃんと0.08[m]の足上げができている。
[image]

Yusuke-Yamasaki-555, 2023-08-03:
だから[このissueの話](https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/issues/6#issuecomment-1661919998)も、間違っているのでは。正弦波の振り幅で0.08を与えていて、それを中腰状態の腰高さから引いているのだから、ちゃんと0.08[m]足を上げなきゃおかしい。

まじでよくわからん。
正しければ、膝下ぐらいまで足首関節の位置が上がっていてもおかしくないのだが。

Yusuke-Yamasaki-555, 2023-08-03:
IKを解析的手法から数値的手法に変更してもいい（いずれそうするつもり）のだが、それしか無いかなぁ。わからん。

[14 more replies]

closed 2023-08-15 by Yusuke-Yamasaki-555