open-rdc/ROS2_Walking_Pattern_Generator #18: Webots内のロボットの動作周期が時々狂う問題

Yusuke-Yamasaki-555, 2023-04-09:
一定周期で歩行パターンをPubして、数回の通信を挟んだ先でWebotsのロボットに動作命令を送っている。
他の状況とかは、[静歩行のissue](https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/issues/17#issue-1658722051)を参照。
以上の状態からして、パッケージ内での通信に原因がありそう。なので、まずはQoSの設定から検証（実験）していく。

以下、参考動作動画。動画の途中で倒れるのだが、その転倒の原因が今回の問題点。

https://user-images.githubusercontent.com/91410662/230777106-bfbd138a-6638-4504-9770-f054ecc0baa6.mp4

Yusuke-Yamasaki-555, 2023-04-09:
[commit 156](https://github.com/open-rdc/ROS2_Walking_Pattern_Generator/issues/17#issue-1658722051)のQoS設定。特に変化なし。問題未だ健在。

```cpp
  static const rmw_qos_profile_t custom_qos_profile =
  {
    RMW_QOS_POLICY_HISTORY_KEEP_LAST,  // History: keep_last or keep_all
    1,  // History(keep_last) Depth
    RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT,  // Reliability: best_effort or reliable
    RMW_QOS_POLICY_DURABILITY_VOLATILE,  // Durability: transient_local or volatile
    RMW_QOS_DEADLINE_DEFAULT,  // Deadline: default or number
    RMW_QOS_LIFESPAN_DEFAULT,  // Lifespan: default or number
    RMW_QOS_POLICY_LIVELINESS_SYSTEM_DEFAULT,  // Liveliness: automatic or manual_by_topic
    RMW_QOS_LIVELINESS_LEASE_DURATION_DEFAULT,  // Liveliness_LeaseDuration: default or number
    false  // avoid_ros_namespace_conventions
  };
```

yasuohayashibara, 2023-04-10:
rosbagを取得して，データが一定周期で送信されているかをチェックすると良いと思います．

Yusuke-Yamasaki-555, 2023-04-10:
ありがとうございます。確認してみます。

Yusuke-Yamasaki-555, 2023-04-10:
ros2 bagで記録を取ってみた。topicしか記録できないので/WalkingPatternの周期だけ確認できた。

`ros2 bag play` 中に `ros2 topic hz /WalkingPattern`（/WalkingPatternのcreate_wall_timerは、600ms）

```bash
average rate: 1.668
	min: 0.599s max: 0.600s std dev: 0.00038s window: 3
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00046s window: 5
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00043s window: 7
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00038s window: 9
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00035s window: 11
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00043s window: 13
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00040s window: 15
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00040s window: 17
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00038s window: 19
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00044s window: 21
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00042s window: 23
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00041s window: 25
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00041s window: 27
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00039s window: 29
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00042s window: 31
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00042s window: 33
average rate: 1.667
	min: 0.599s max: 0.600s std dev: 0.00042s window: 35
```

/WalkingPatternのPublish周期は問題なかった。他のService周りが問題？それか標準出力を大量にさせているから？

Yusuke-Yamasaki-555, 2023-04-10:
標準出力は関係なさそう。

次は、Serviceごとの動作周期を、get_logger()の標準出力から見てみようと思う。

Yusuke-Yamasaki-555, 2023-04-11:
ServiceのReq/ResごとにRCLCPP_INFOで出力。出力された時間の差分を見て判断。（もっとスマートな方法もあるが、即興）

Serviceのチェックリスト
- [ ] WalkingStabilizationController <=> WalkingPatternGenerator
- [ ] WalkingStabilizationController <=> Kinematics (FK)
- [ ] WalkingPatternGenerator <=> Kinematics (IK)

Yusuke-Yamasaki-555, 2023-04-11:
WalkingStabilizationController, WalkingPatternGenerator間のServiceを確認。以下、その記録。

第１値：前回Response〜今回Response間の経過時間
第２値：第１値の最大経過時間
第３値：第１値の最小経過時間

結果として、転倒した（周期が狂った）時に、Response間の経過時間が大きくなる周期と、経過時間が小さくなる周期が連続していることが見て取れた。ただ、転倒時にそのような出力が見られた回数は少ない。十数回記録したが、経過時間に明らかに現れたのは数回（４、５回）だけだった。体感、３，４割。
この結果から、通信周りに原因がある可能性が考えられる。

---
1
[driver-5] [WebotsRobotHandler]: 0.0369551    max: 0.0427454    min: 0.0243974
[driver-5] [WebotsRobotHandler]: 0.0329161    max: 0.0427454    min: 0.0243974
[driver-5] [WebotsRobotHandler]: 0.0305052    max: 0.0427454    min: 0.0243974
[driver-5] [WebotsRobotHandler]: 0.0290225    max: 0.0427454    min: 0.0243974
[driver-5] [WebotsRobotHandler]: 0.0641863    max: 0.0641863    min: 0.0243974
[driver-5] [WebotsRobotHandler]: 0.00552893    max: 0.0641863    min: 0.00552893
[driver-5] [WebotsRobotHandler]: 0.0392299    max: 0.0641863    min: 0.00552893
[driver-5] [WebotsRobotHandler]: 0.0321083    max: 0.0641863    min: 0.00552893
---
2
[driver-5] [WebotsRobotHandler]: 0.0317235    max: 0.0422983    min: 0.0233049
[driver-5] [WebotsRobotHandler]: 0.0366919    max: 0.0422983    min: 0.0233049
[driver-5] [WebotsRobotHandler]: 0.0507896    max: 0.0507896    min: 0.0233049
[driver-5] [WebotsRobotHandler]: 0.0132678    max: 0.0507896    min: 0.0132678
[driver-5] [WebotsRobotHandler]: 0.0321977    max: 0.0507896    min: 0.0132678
[driver-5] [WebotsRobotHandler]: 0.033932    max: 0.0507896    min: 0.0132678
---
3
[driver-5] [WebotsRobotHandler]: 0.0289981    max: 0.0428576    min: 0.0218713
[driver-5] [WebotsRobotHandler]: 0.0362425    max: 0.0428576    min: 0.0218713
[driver-5] [WebotsRobotHandler]: 0.0334954    max: 0.0428576    min: 0.0218713
[driver-5] [WebotsRobotHandler]: 0.0511029    max: 0.0511029    min: 0.0218713
[driver-5] [WebotsRobotHandler]: 0.0167382    max: 0.0511029    min: 0.0167382
[driver-5] [WebotsRobotHandler]: 0.0325301    max: 0.0511029    min: 0.0167382
[driver-5] [WebotsRobotHandler]: 0.0316057    max: 0.0511029    min: 0.0167382
[driver-5] [WebotsRobotHandler]: 0.0294566    max: 0.0511029    min: 0.0167382

Yusuke-Yamasaki-555, 2023-04-11:
WalkingStabilizationController <=> Kinematics (FK) 間のServiceを確認。手法は上記WalkingStabilizationController<=>WebotsRobotHandlerに同じ。

結果として、こちらでも転倒時に経過時間（動作周期）が大きくなる周期と小さくなる周期が連続していることが分かった。また、前実験同様、転倒しても周期の変化が確認できないときがあった。ただ、前実験よりも確認できた割合が増加したように感じる（記録回数を数えとけば良かった...）。
結果から、どうもKinematics周りが原因な気がしている。これでWalkingPatternGeneratorでも同様の結果が見られた場合、Kinematicsへのアクセスで競合が起きている可能性が考えられる。

1
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.03194    max: 0.0434449    min: 0.0217736
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0324066    max: 0.0434449    min: 0.0217736
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0535579    max: 0.0535579    min: 0.0217736
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0128713    max: 0.0535579    min: 0.0128713
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0318248    max: 0.0535579    min: 0.0128713
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0341432    max: 0.0535579    min: 0.0128713

2
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0296953    max: 0.0390635    min: 0.024018
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0321951    max: 0.0390635    min: 0.024018
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0317268    max: 0.0390635    min: 0.024018
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0505967    max: 0.0505967    min: 0.024018
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0100808    max: 0.0505967    min: 0.0100808
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0341976    max: 0.0505967    min: 0.0100808
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0374908    max: 0.0505967    min: 0.0100808

3
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0345831    max: 0.0433075    min: 0.0231047
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0346551    max: 0.0433075    min: 0.0231047
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0250373    max: 0.0433075    min: 0.0231047
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0578792    max: 0.0578792    min: 0.0231047
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0128949    max: 0.0578792    min: 0.0128949
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0356398    max: 0.0578792    min: 0.0128949
[walking_stabilization_controller-3] [WalkingStabilizationController]: 0.0363894    max: 0.0578792    min: 0.0128949

Yusuke-Yamasaki-555, 2023-04-11:
WalkingPatternGenerator <=> Kinematics (IK) 間のServiceを確認。手法はいままでと同様。ただ、ここだけ周期が600msで固定（create_wall_timer）されていることに注意。

結果として、今まで見られていた、動作周期が長い周期と短い周期が連続している出力は見られなかった。それも当然で、ros2 bag での topic /WalkingPattern の出力周期が一定だったことからも、FKとのService通信に乱れがないことは既に確認されている。ので、出力記録は示さない（示す必要は無いだろう）。

Yusuke-Yamasaki-555, 2023-04-11:
以上の結果から、問題はKinematicsのFKにある可能性が考えられる。

まずはFKとの通信を切ってみて、歩行が安定するか確認してみる。

Yusuke-Yamasaki-555, 2023-04-11:
FKを切ってみた結果、動作周期は乱れることなく、安定した周期で静歩行ができた。
原因は、FK本体かFKとの通信周り、あるいはWSCにあると考えられる。WSCに原因がある場合、Service通信のServer内で別のService通信を行うことが原因となっている気がする。可能性は低いだろうし、気がするだけ。
なぜ毎周期ではなく、ランダムなタイミングで周期が狂うのかは不明。タイミングがランダムかすら分からん（調べていない）。

FKを切ると、転倒することなく８分超の静歩行を続けることができた。

[image]

Yusuke-Yamasaki-555, 2023-04-11:
後回しで。

先にWalkingPatternGeneratorを開発する。

Yusuke-Yamasaki-555, 2023-04-11:
記録
Kinamticsを挟まなくても、WRH2WSCのServiceの処理時間は以下。参考程度に。動歩行になると影響が大きくなる可能性がある。

[driver-5] [WebotsRobotHandler]: 0.0341406    max: 0.066509    min: 0.0232551
[driver-5] [WebotsRobotHandler]: 0.0301347    max: 0.066509    min: 0.0232551
[driver-5] [WebotsRobotHandler]: 0.0355928    max: 0.066509    min: 0.0232551
[driver-5] [WebotsRobotHandler]: 0.0299053    max: 0.066509    min: 0.0232551

Yusuke-Yamasaki-555, 2023-04-19:
処理の流れ：

<1>
WalkingPatternGenerator
->service request（非同期）
IK
->service response（非同期）
WalkingPatternGenerator
->publish ----- 600[ms]周期でpublish -> IKでの処理時間の増減はpublishには影響していない。
->subscribe
WalkingStabilizationController

<2>
WebotsRobotHandler
->service request（非同期）
WalkingStabilizationController
->service request（非同期）
IK
->service response（非同期）
WalkingStabilizationController
->service response（非同期）
WebotsRobotHandler
->ROBOTIS OP2へ適用（現状、WalkingStabilizationControllerでの処理を反映していない）

<1>、<2>を並行して実行。（意識はしてなかったが、並行処理になっていると思う）

WalkingPatternGeneratorが歩行パターンをWalkingStabilizationControllerにPub。
WebotsRobotHandlerがWalkingStabilizationControllerから歩行パターンをResponse。

現状調査まとめ：
* WalkingStabilizationControllerからIKへのService通信は、非同期
* IKへService通信は、WalkingPatternGeneratorとWalkingStabilizationControllerがおこなっている。Requestタイミングの調整は施していない。
  * WalkingStabilizationControllerがRequestする相手がIKではなくFKでも、同様の症状が見られている。WalkingPatternGeneratorとWalkingStabilizationControllerの競合以外の原因が考えられる。
* Service通信周りにも、根本的な原因はなさそう。以下、IK内部での処理時間をRequest毎に記録したものの一部。転倒し終わったタイミングでの出力結果。maxが最長処理時間、minが最短処理時間。単位は[sec]。maxとminの差が大きく、処理時間が一定でないことがわかる。（0.1[ms]ほどの話ではあるが）

[ik_srv-2] [FK]: 0.000157356    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 0.000603199    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 0.000453234    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 0.000452518    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 0.000231981    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 0.000565052    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 0.000120163    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 6.03199e-05    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 0.000579357    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 0.000707865    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 0.000568867    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 0.000537157    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 0.000449181    max: 0.00105429    min: 5.88894e-05
[ik_srv-2] [FK]: 0.000542402    max: 0.00105429    min: 5.88894e-05

* ここでの処理時間の変化が、直接ロボットの動作に影響を与えている可能性が高い。実際、Webotsのstepの周期を固定していないので、WebotsRobotHandler -> WalkingStabilizationController -> IK -> WalkingStabilizationController -> WebotsRobotHandler での通信時間を含めた処理時間が直接Webotsの1stepの時間となっているはず。
* IKでの動作時間の差が、Service通信を通すことによって増幅され、1stepの時間に影響を与えている可能性。WalkingStabilizationControllerでも処理時間の増減が見られる。IKでの処理時間の増減と連動しているかは未調査。

closed 2023-10-23 by Yusuke-Yamasaki-555