open-rdc/ROS2_Walking_Pattern_Generator #26: ソフトウェアアーキテクチャの考え直し、実装

Yusuke-Yamasaki-555, 2023-04-25:
・１つのnodeで、Topic通信とService通信で競合が起きている可能性、歩行パターンのPublishとWebotsのstepの大きさの不一致から来る何らかの不具合の可能性
　ー＞　考えられる問題はすべてコード云々ではなく、ソフトウェアアーキテクチャに問題があると判断。もし違ったとしても、アーキテクチャからくる可能性は除去できる（競合と不一致を解消すれば）。

＝＞　ソフトウェアのアーキテクチャを考え直し、改めて適用する。

Yusuke-Yamasaki-555, 2023-04-26:
ソフトウェアアーキテクチャ第２案

[image]

Yusuke-Yamasaki-555, 2023-04-30:
ik, fkで分ける必要はなく、kinematicsとして1つの共有ライブラリにまとめたほうが綺麗
他、ヤコビ行列の計算やstd::array2eigen系の関数は、matrix_calculation的な共有ライブラリ名で、別でまとめようか

Yusuke-Yamasaki-555, 2023-05-05:
sync service 通信にかかる時間を計測したほうが良い。
いままでの測定方法は、間にstd::cout << std::endlを入れていたりと、関係ない処理(しかもクソ重い処理を！)が測定時間に含まれてしまっていた可能性がある(過去のcommitからコードを要確認)。何も考えていなかった。
今度からは、既に有志から公開されているデータを参考にしたり、間に無駄な処理を挟まないべき。出力もendlとかは避けて、vecに値を貯めて最後にprintfするのがマシかな？後はboot型でいいのを探すとか。

以下、良さそうなサイト
https://research.reazon.jp/blog/2023-01-15-DDS-performance.html

https://www.eprosima.com/index.php/resources-all/performance/eprosima-fast-dds-performance

https://discourse.ros.org/t/how-to-track-dds-performance-in-ros-2/22498

https://eprosima.com/index.php/products-all/eprosima-fast-dds-monitor

Yusuke-Yamasaki-555, 2023-05-05:
全体で、1stepを、10ms以内に収めることを目標とする！Brainsは10msらしい。参考にする。

yasuohayashibara, 2023-05-05:
> 間にstd::cout << std::endlを入れていたり

なるほど．今まで処理が周期的に実行できていなかったのは，これが原因かもしれません．
標準出力は不定期に長い時間を要する処理を割り込ませます．

以前のプログラムでこの標準出力を削除して実行してみてください．
解決はしないかもしれませんが，少なくとも改善はすると思います．
ファイルの書き出しはそれほど時間がかからないことが多いので，もしログを残したい場合は標準出力を使用するのではなく，ファイルに書き出すと良いです．

Yusuke-Yamasaki-555, 2023-05-09:
ご指摘していただき、ありがとうございます。
ご指摘の方法を試して見たのですが、やはり転倒してしまいますので、直接的な原因では無いようです。
今まで転倒するタイミングを測定していませんので改善の有無を判断することはできないのですが、数回ほど試してみても改善した感じはありませんでした。

しかし、Debug用に標準出力を多様してしまっていたのは確かで、他issueでの通信時間にも標準出力の処理時間が多く含まれてしまっている可能性が大きいです。ここに関しましては、以前に井上先輩からも指摘していただきましたので、新しいアーキテクチャには反映する予定で考えています。

Yusuke-Yamasaki-555, 2023-05-10:
（ちゃんと専門用語を用いて説明できたほうが良い。でないと、相手に全く伝わらない。）
（Ver1の問題点に関して、原因をある程度絞れていなければソフトウェアアーキテクチャは改善できない。今回の場合は、ある程度まで原因を絞れていたが、そこを解決するのではなく全てをやり直した。本筋ではないからそれでも良いが、今後同じようなことが起きたときは、その原因を解決するように動いたほうがいい場合もあるだろう。）

Yusuke-Yamasaki-555, 2023-05-10:
callcack_group を使う、加えてserviceを複数呼び出す場合は、MultiThreadedExecutorが必要っぽい。
それ以前に、この問題はCompornent指向の実装を行えば済む話なのかもしれない。いや、futureで待つことは必要なんだろうな。

https://answers.ros.org/question/343279/ros2-how-to-implement-a-sync-service-client-in-a-node/

Compornent同士でのservice通信において、非同期と同期それぞれの挙動を調査するべき。同期にするプログラムにしなかったときに、request直後のプログラムがresponse前に実装されるかどうかを確認すれば良いのかも。

Yusuke-Yamasaki-555, 2023-05-11:
referense

https://gbiggs.github.io/rosjp_ros2_intro/ros2_basics.html#%E3%82%B5%E3%83%BC%E3%83%93%E3%82%B9%E3%81%AE%E5%AE%9A%E7%BE%A9%E3%81%A8%E5%88%A9%E7%94%A8

https://qiita.com/NeK/items/aadb89da577e3706e674

Yusuke-Yamasaki-555, 2023-05-30:
右往左往しながら、ソフトウェアアーキテクチャ案のVer.4を考えた。Ver.3もあるが、それを少し改良したもの。

Ver.3

[image]

Ver.4

[image]

Yusuke-Yamasaki-555, 2023-05-30:
で、問題の１つとして残っていた同期Service通信の中で別の同期Service通信を行うと、処理が止まってしまう問題が、どうも解決できそうだ。

Ver.4を想定して、create_wall_timerで一定周期で回す関数の中で、service通信を同期（future.wait_for(?s) でresを待つことで擬似的に同期にしている）でおこなうことができた。
[公式にあるコード](https://docs.ros.org/en/humble/How-To-Guides/Using-callback-groups.html)を参考にすれば行けた。

肝は、
* 複数のspinを有するnodeは、MultiThreadExecutorでマルチスレッドで動かす。
* CallbackGroupを使って、同一の相互排他的CallbackGroup（デフォルト）でCallbackを回すことを回避する。多分、１つのスレッドでは１つのspinしか回せないから、マルチスレッドにしてCallbackGroupを分ければOKってことな気がする。
* main関数でnodeをspinさせるとき、add_nodeやspinに渡すnodeは、一旦、別変数にmake_sharedで渡して、その別変数をadd_nodeやspinに渡してやる。何故かココができていないのも原因の一つだった。

ってな感じ。

この解決方法は、同期Service通信の中で別の同期Service通信を回そうとしたときにも有効だと思われるので、試してみたほうがいい。

Yusuke-Yamasaki-555, 2023-05-30:
以下、参考サイト
https://github.com/ros2/rclcpp/issues/206
https://answers.ros.org/question/354132/how-to-call-spin_some-or-spin_until_future_complete-in-member-functions/
https://www.youtalk.jp/2017/06/16/rclcpp2.html
https://answers.ros.org/question/343279/ros2-how-to-implement-a-sync-service-client-in-a-node/
https://docs.ros.org/en/humble/How-To-Guides/Using-callback-groups.html
https://github.com/ros2/rclcpp/issues/312
https://github.com/ros2/rclcpp/issues/206

Yusuke-Yamasaki-555, 2023-06-13:
[image]

　アーキテクチャVer4の、Topic１と３以外を結合して動かしてみた。マネージャのPublishは仕様通りの10[ms]。
　結果、成功した。が、SubscribeであるコントローラがマネージャからのTopicを、毎週期、連続で拾えないこともわかった。ほとんどの場合は連続で拾えているのだが、たまに１つのTopicを受け取れないことがあった。
　要するに、10[ms]周期で出力される情報を拾いそこねることがたまにあるということ。ただ、２つ以上のTopicを連続して拾い損ねることは確認されていない。
　原因はWebots側の basicTimeStepが制御周期と近い値であることが原因であると思われる。1[ms]とかで回せば良いのだろうか。ただ、それはベーシックではないなと思う。RoboCupのWebots環境を参考にできるだろう。確か、waitをしていたはず。
　この問題は歩行パターンを拾いそこねるということなので、非常に良くない。明らかに同期通信ができたほうが良い箇所。

★マネージャで採用しているプログラムのように、同期Serviceに変えてVer3を採用するか？
　Topic通信のままにして、歩行パターンが更新されるまでWaitできるならするか？
　
　それか、もう強引に、webots_ros2を使わずに、Webotsを直接使えるようにするか？WebotsとROS 2の絡みにwebots_ros2を使わずに、WebotsのプログラムをROS 2パッケージに含めるようにしてみるか？

Yusuke-Yamasaki-555, 2023-06-13:
ソフトウェアアーキテクチャは一旦おやすみ。動歩行の実装にチャレンジしていく。

動歩行の実装は、コントローラの中に直接関数として実装する予定。もちろん後々にアーキテクチャがイケたら他Nodeに移植する。

Yusuke-Yamasaki-555, 2023-08-17:
architecture ver.??

[image]

Yusuke-Yamasaki-555, 2023-08-26:
### メモ

　現状、情報の可視化とかRviz2, Gazeboへの対応を色々と考える中で、ros2_controlとかMoveit!2といった大規模なROS 2のアプリケーションの採用をどうするか考えている。

　しかし今回のヒューマノイドロボットの歩行制御ソフトウェアは、できる限りの汎用性・拡張性・可読性を持たせたい、加えて使用・利用のハードルをできる限り下げたい。そう考えると、大規模でかつROS 2専用のアプリケーションには、あまり依存したくない。
　理由としては、
　* ソフトウェア全体の大黒柱になってしまうことで、他ミドルウェアへの移行が難しくなる恐れ
　* 大規模であることから、事前知識や内部のアーキテクチャへの理解を前提としたソフトウェアになってしまう恐れ
が主に挙げられる。

　とにかく、依存するものを極力最小限に抑えたい。

　もちろん利点も大きい。既存のアプリケーションやライブラリを採用する利点として、高い信頼性の確保と可読性の向上、コストの削減が主に挙げられる。
　だからといって無闇矢鱈に採用していては、アーキテクチャの複雑化や外部への高い依存性を持ってしまい、拡張性やメンテナンス性が死んでしまう。加えて依存先と生命を共にすることになりかねない。
　高い信頼性の確保が特に必要な箇所はどこであるかを考えるべきで、外部への依存もできる限り小さく収めたほうがより拡張性やメンテナンス性を確保できる。

　だからもしMoveit!2やros2_controlを使うにしても採用範囲は最小限に抑えたいわけで。
　この２つを使うメリットはハードウェア側を強く抽象化できることだと思う。あとRviz2などでのモニタリングにも有効に働く。

　なので採用するならRobotHandler nodeだけかな。制御理論を含んだManager側では全く依存せずに、Manager <-> Handler間で動作命令＆Feedbackをやり取りする感じで。

Yusuke-Yamasaki-555, 2023-08-26:
全ての要素がトレードオフでやゔぁい。

Yusuke-Yamasaki-555, 2023-09-06:
Architecture Ver.???

この組み方、結構良いかも。アーキテクチャは他人の評価が欲しくなるな。

[image]

yasuohayashibara, 2023-09-07:
口頭で質問するか，もう少し整理した図をいただければコメントできます．

Yusuke-Yamasaki-555, 2023-09-13:
Arch Ver.???
趣向を変えてみた。

[image]

Yusuke-Yamasaki-555, 2023-09-14:
公式 pluginlibの話
https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Pluginlib.html

 通信のオーバーヘッドが怖い、topic lossが怖いなら、全部plugin化すれば良いじゃない。て感じ。
 で、並列化できる所はガンガンしていく。

 歩行パターンジェネレータと歩行安定化制御系は、configファイルからユーザが選べるようにしたい。歩行安定化制御系は、利用の有無も選べるようにしたい(選ぶ設定にnoneと書いたら使わない判定)。
 で、既存で1つ以上用意しておいて、ユーザ側でも自作できるようにしたい。
 
 外とのtopicはstandard messageで。

Yusuke-Yamasaki-555, 2023-09-14:
目指すのは、
**パッケージをインストールして、各設定をすれば、歩行制御の中身を意識する事なく歩行できること**。

WPG, WSCをplugin化するわけだけれども、IK周り(歩行パターンから足関節角度・角速度への変換)もplugin化するべきか。ハード特性によって異なるしheuristicな修正方法も必要なわけだから、取っ替えやすくするべき。
それか、読み込んだURDFから動的に変化させる。ただこれだと拡張性が死ぬ気がするんだ。

Yusuke-Yamasaki-555, 2023-09-19:
ソフトウェアアーキテクチャの決定

以下のアーキテクチャで、以降はひとまず進めていく。
[image]

architecture branchで移行を進める。

Yusuke-Yamasaki-555, 2023-09-19:
Debug modeでのTopicのPublishは、Pluginに処理を投げているときに並列処理としてあるのもありだと考えている。

Yusuke-Yamasaki-555, 2023-10-12:
メモ
* Cmakeで各々のPackageが用いるC++のVersionを合わせるべき。C++17で良いかな。Versionごとの違いはあまり把握できていないが、スマートポインタ周りの仕様的にC++17以上が良いかもしれない。特に重要ではないだろう。
* Classのデストラクタもちゃんと把握して明示するべきかもしれない。

Yusuke-Yamasaki-555, 2023-10-18:
OKOKOK、アーキテクチャ移行８割がた完了した。移行前と同様の歩行が実現できた。
動画は撮らずに省略。

Yusuke-Yamasaki-555, 2023-10-18:
あとは、
* Loggerの実装
* Debug Modeの実装
* Rviz2周りの統合とROBOTIS OP2のURDFのLicense整備
* WalkingStabilizationControllerとConvertToJointStatesを修正
* README.mdの整備
が済めば、mainブランチにプルリクして、必要最低限は済む感じになるかな。

追加として、
* 汎用性確保のための、URDFやテキストからParameterを読み取るScriptの開発
* 汎用性確保のための、ソフトウェアの改善
* オンライン歩行パターン生成への対応
* 必要なLogをRviz2やGUIに表示
* Wikiに使用方法や拡張方法など、ソフトウェアの全仕様を記載
ができれば、80点。
ここまでできれば、実験もできるようになるだろう。

Yusuke-Yamasaki-555, 2023-10-18:
URDFからProtoへ変換して、そのProtoをWorldファイルに自動で適用してくれるScriptがほぢい。これが作れれば実験や汎用性が超絶進歩する。

yasuohayashibara, 2023-10-19:
こちらはいかがでしょうか？
https://github.com/cyberbotics/urdf2webots

Yusuke-Yamasaki-555, 2023-10-19:
ありがとうございます。丁度そちらを使おうと考えていました。コレを含めた自動化スクリプトを書こうと考えています。

Yusuke-Yamasaki-555, 2023-10-22:
WSCとCTJSを毎Stepごとに実行するようにした。

　ただ、ConvertToJointStates Pluginのレイテンシが、約 7 ~ 0.5 [ms] もの振り幅がある。特に最大値がちと厳しくて10 [ms]で回すことを考えると改善したいところ。
　ポインタを多用してアドレスにアクセスするレイテンシが結構重いのだろうか。それとも潜在的なバグか、IK（解析的手法）か。
　まぁコード自体、冗長なものになっているかもしれないので、全体的に見直す必要があるだろう。

Yusuke-Yamasaki-555, 2023-10-23:
CtJSの実行時間が安定にできたら、このIssueも閉じれるだろう。

で、今までLoggingはFile出力でやっていたが、iostreamを使うのは実行時間が未知になったりするらしいので、実行時間計測のLogとか諸々は、Publishで他Nodeに任せる or Threadsで並行処理させることにしたい。

Yusuke-Yamasaki-555, 2023-10-23:
一応最低限度のものができたので、コードのスタイルも少しずつ意識していきたい。
特にPluginの返り値がポインタ渡しなのを参照渡しにしたりとか

Yusuke-Yamasaki-555, 2024-01-26:
アーキテクチャは定まって実装完了した。ので、Close。
各プラグインとRobot Managerの実行時間は他に示す予定。

closed 2024-01-26 by Yusuke-Yamasaki-555