open-rdc/ROS2_Walking_Pattern_Generator #36: ros2 launch robot_bringup robot_bringup_launch.py で出力されたエラー

tomoki057, 2024-09-26:
[INFO] [launch]: All log files can be found below /home/tomo0724/.ros/log/2024-09-26-22-41-47-959647-satomo724pc-31513
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - PackageNotFoundError: "package 'robot_recorder' not found, searching: ['/home/tomo0724/ros2_ws/install/robot_bringup', '/home/tomo0724/ros2_ws/install/robot_tools', '/home/tomo0724/ros2_ws/install/robot_description', '/home/tomo0724/ros2_ws/install/mypkg', '/opt/ros/humble']"
 - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown

Yusuke-Yamasaki-555, 2024-09-26:
ほうほう

```
sudo apt install hogehoge
```

Yusuke-Yamasaki-555, 2024-09-26:
おそらく、@tomoki057くんの環境が悪いかもしれない。
なので、このissueはclose

Yusuke-Yamasaki-555, 2024-09-26:
問題が解決していないので、一応openにしておこう。

closed 2025-04-11 by Yusuke-Yamasaki-555